akaJes/marlin-config #1: Installation - windows

christianh17, 2017-05-28:
Hi,
I would love to try your config-tool. But: I have no idea at all how to install it with windows. I do have linux machines, but none with a graphical UI - only accesible via putty or of course webinterfaces.

Can you please give me a hint?
Regards
Christian

akaJes, 2017-05-28:
Hi
this [Windows-Installation](https://github.com/akaJes/marlin-config/wiki/Windows-Installation) can help for you
if you do not have GUI - it is not problem :) - you can enter from another device - just change localhost to your machine name
```
Marlin config tooll started on port http://localhost:3000
```

christianh17, 2017-05-28:
Hi!
Thanks, it works now. It was so easy - in German we say: I had tomatoes on my eyes.
It is totally different compared to the configuration I did before, looks much easier. I will try it!
Thanks a lot for this great part of software,
Christian

akaJes, 2017-05-28:
Thank you too
You can install PlatformIO as optional tool - it allow to compile and upload firmware

iain1965, 2017-06-03:
when trying to install on windows 10, I get npm is not defined error, and pio is not defined error

iain1965, 2017-06-03:
why can't we just have a straightforward windows installer, and a windows gui...?

akaJes, 2017-06-03:
https://github.com/akaJes/marlin-config/releases/tag/2.6.1x
dependency is git https://git-scm.com/downloads