akaJes/marlin-config #4: Compiling issues updateting issues

HaasNL2, 2017-07-10:
So I like the idea of this tool very much, and I am trying to work on it on a windows pc. And I am having different results.
I now twice got it working with 1.1.4, but I can only get to 1.1.4 if I do a complete new firmware.
If I used a previous save (save always opens with 1.1.3) I get this warning:

> Oh snap!
> error: Your local changes to the following files would be overwritten by checkout: Marlin/Configuration.h Marlin/Configuration_adv.h Please commit your changes or stash them before you switch branches. Aborting 

I did not even change anything from when I started. Last time I had it up and running I left my pc on the entire weekend :). So when it worked it worked good.
1.1.3 will not compile I get a lot of errors:

> 
> Verbose mode can be enabled via `-v, --verbose` option
> ValueError: No escaped character:
> File "C:\python27\lib\lib\site-packages\platformio\builder\main.py", line 155:
> env.SConscript("$BUILD_SCRIPT")
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
> return _SConscript(self.fs, *files, **subst_kw)
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
> exec _file_ in call_stack[-1].globals
> File "C:\Users\Me\.platformio\platforms\atmelavr\builder\main.py", line 217:
> target_elf = env.BuildProgram()
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
> return self.method(*nargs, **kwargs)
> File "C:\python27\lib\lib\site-packages\platformio\builder\tools\platformio.py", line 58:
> env.ProcessFlags(env.get("BUILD_FLAGS"))
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
> return self.method(*nargs, **kwargs)
> File "C:\python27\lib\lib\site-packages\platformio\builder\tools\platformio.py", line 116:
> parsed_flags = env.ParseFlags(str(flags))
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 800:
> do_parse(arg)
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 706:
> params = shlex.split(arg)
> File "c:\python27\lib\lib\shlex.py", line 279:
> return list(lex)
> File "c:\python27\lib\lib\shlex.py", line 269:
> token = self.get_token()
> File "c:\python27\lib\lib\shlex.py", line 96:
> raw = self.read_token()
> File "c:\python27\lib\lib\shlex.py", line 191:
> raise ValueError, "No escaped character"
>  [ERROR] Took 0.75 seconds 
> 
> 
> 
> Verbose mode can be enabled via `-v, --verbose` option
> ValueError: No escaped character:
> File "C:\python27\lib\lib\site-packages\platformio\builder\main.py", line 155:
> env.SConscript("$BUILD_SCRIPT")
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
> return _SConscript(self.fs, *files, **subst_kw)
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
> exec _file_ in call_stack[-1].globals
> File "C:\Users\Me\.platformio\platforms\atmelavr\builder\main.py", line 217:
> target_elf = env.BuildProgram()
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
> return self.method(*nargs, **kwargs)
> File "C:\python27\lib\lib\site-packages\platformio\builder\tools\platformio.py", line 58:
> env.ProcessFlags(env.get("BUILD_FLAGS"))
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
> return self.method(*nargs, **kwargs)
> File "C:\python27\lib\lib\site-packages\platformio\builder\tools\platformio.py", line 116:
> parsed_flags = env.ParseFlags(str(flags))
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 800:
> do_parse(arg)
> File "C:\Users\Me\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 706:
> params = shlex.split(arg)
> File "c:\python27\lib\lib\shlex.py", line 279:
> return list(lex)
> File "c:\python27\lib\lib\shlex.py", line 269:
> token = self.get_token()
> File "c:\python27\lib\lib\shlex.py", line 96:
> raw = self.read_token()
> File "c:\python27\lib\lib\shlex.py", line 191:
> raise ValueError, "No escaped character"
>  [ERROR] Took 0.75 seconds 
> 
All dependency are running nice. I am using windows 10 pro.
>

akaJes, 2017-07-10:
hi
did you install marlin-conf from exe file? or through npm package? and what version installed? which marlin release?
about first issue:
1. i just save my Configuration* files to safe place 
2. select 'reset' from menu
3. select  another version
4. drag in the browser my saved Configuration files

about second issue:
it looks like platform.io issue
did you install last version of python? and platform.io?
you can check it from Marlin folder - where placed platformio.ini file type `pio run`,
if not try to remove `.pioenvs` `.piolibdeps` cache folder
ill try to deploy this case on windows 10 tomorrow

HaasNL2, 2017-07-10:
Here is my log if I run pio run
`Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_64128n.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_dogm128.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_dogm132.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_lm6059.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_lm6063.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_nhd_c12832.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7565_nhd_c12864.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7687_c144mvgd.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7920_128x64.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7920_192x32.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_st7920_202x32.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_t6963_128x128.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_t6963_128x64.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_t6963_240x128.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_t6963_240x64.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_tls8204_84x48.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1601_c128032.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1608_240x128.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1608_240x64.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1610_dogxl160.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1611_dogm240.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1611_dogxl240.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1701_dogs102.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_dev_uc1701_mini12864.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_ellipse.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_font.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_font_data.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_line.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_ll_api.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_page.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb14v1.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb16h1.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb16h2.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb16v1.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb16v2.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb32h1.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb8h1.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb8h1f.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb8h2.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb8h8.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb8v1.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pb8v2.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pbxh16.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_pbxh24.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_polygon.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_rect.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_rot.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_scale.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_state.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_u16toa.o
Compiling .pioenvs\mega2560\src\Marlin - kopie\Marlin\.piolibdeps\U8glib_ID7\src\clib\u8g_u8toa.o
[... cut ...]
*** [.pioenvs\mega2560\src\Marlin - kopie\Marlin\Sd2Card.o] Error 1
*** [.pioenvs\mega2560\src\Marlin - kopie\Marlin\Marlin_main.o] Error 1
*** [.pioenvs\mega2560\src\Marlin - kopie\Marlin\SdBaseFile.o] Error 1
================================================================================================================================================ [ERROR] Took 22.82 seconds ================================================================================================================================================

C:\Users\ME\Desktop\marlin>`

HaasNL2, 2017-07-10:
Wierd I noticed the long folder structure, so i tried to it on the c:\ root. And it worked.
Fresh download worked then i put my old config in it, and it still worked.
Maybe that was the issue?

akaJes, 2017-07-11:
it seems like some  file access restriction  or corrupted file issue

HaasNL2, 2017-07-13:
About that acces issue:
I get this error when booting from administrator cmd.
```

C:\>mct
Node v6.10.3
fatal: Not a git repository (or any of the parent directories): .git

no root
(node:16036) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): fatal: Not a git repository (or any of the parent directories): .git

press Enter to clone it from web or ^C to exit(node:16036) PromiseRejectionHandledWarning: Promise rejection was handled asynchronously (rejection id: 1)
```

akaJes, 2017-07-14:
you no need to use mct with administrative rights, only with user rights and in users' home folder
if you run single time with administrative rights - files will be locked for user rights