akaJes/marlin-config #6: mac/platformio

mitchy2233, 2017-07-16:
its installed and up to date yet when i try to compile the firmware it tells me that i need to install platform wtf do i do??????? 

Requirement already up-to-date: platformio in /Library/Python/2.7/site-packages
Requirement already up-to-date: lockfile<0.13,>=0.9.1 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: requests<3,>=2.4.0 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: bottle<0.13 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: arrow<1 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: click<6,>=5 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: semantic-version>=2.5.0 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: colorama in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial!=3.3,<4,>=3 in /Library/Python/2.7/site-packages (from platformio)
Requirement already up-to-date: urllib3<1.22,>=1.21.1 in /Library/Python/2.7/site-packages (from requests<3,>=2.4.0->platformio)
Requirement already up-to-date: idna<2.6,>=2.5 in /Library/Python/2.7/site-packages (from requests<3,>=2.4.0->platformio)
Requirement already up-to-date: certifi>=2017.4.17 in /Library/Python/2.7/site-packages/certifi-2017.4.17-py2.7.egg (from requests<3,>=2.4.0->platformio)
Requirement already up-to-date: chardet<3.1.0,>=3.0.2 in /Library/Python/2.7/site-packages (from requests<3,>=2.4.0->platformio)
Requirement already up-to-date: python-dateutil in /Library/Python/2.7/site-packages (from arrow<1->platformio)
Requirement already up-to-date: six>=1.5 in /Library/Python/2.7/site-packages/six-1.10.0-py2.7.egg (from python-dateutil->arrow<1->platformio)

mitchy2233, 2017-07-16:
why the hell wouldn't you have the option of using arduino ied??????????????

akaJes, 2017-07-16:
if the buttons compile/flash disabled - you need to restart application
you can use arduino also - just open the same folder from it

ravini-mephisto, 2017-07-16:
Mitch, you're being rude while asking for help.  If you want to use Arduino go right ahead.

Bazeone, 2017-07-17:
I'm also having this issue. I've installed platformio as described in the guilde and it's not seeing it's installed.

akaJes, 2017-07-18:
if you type in console `pio` or `platformio` and get an error - it means that PlatformIO not installed properly 
my application just checks it before - if not - shows poupup

Bazeone, 2017-07-18:
They both work via console

akaJes, 2017-07-18:
what OS do you use? it was tested on ubuntu and windows, i have no mac

Bazeone, 2017-07-18:
i'm on macOS Sierra 10.12.5

ravini-mephisto, 2017-07-19:
This might be related to the pyserial bug in PlatformIO https://github.com/platformio/platformio-core/commit/a37eb9868f3b20e982d0c3cd1a742fcb8ab60efc

I suspect that, because PlatformIO is unable to be called while this bug exists, that's why Marlin-Config doesn't think it is installed.

ravini-mephisto, 2017-07-19:
Downgrade pyserial and see if it works

akaJes, 2017-07-19:
I see another one simple solution:
Try to upgrade to dev version
```
# uninstall existing version
pip uninstall platformio

# install the latest development version of PlatformIO
pip install -U https://github.com/platformio/platformio-core/archive/develop.zip
```

akaJes, 2017-07-20:
I have build test release for macos with the console support and pio fixes

ravini-mephisto, 2017-07-20:
Tested with OS X Sierra and working.

akaJes, 2017-07-20:
Does the console work also?
because i did it on ubuntu under vmware with pc-hackintosh

ravini-mephisto, 2017-07-20:
I can't test it out; at the moment my printer is connected to my linux box.