akaJes/marlin-config #7: Where to donate?

ravini-mephisto, 2017-07-16:
This project is awesome, and I've made a tutorial vid for my user groups on how to use it for updating their firmware https://www.youtube.com/watch?v=hUeMQfPTdac

Where is your donation button?  Get a PayPal.me button or something on your profile page.  I'd like to support this project.

akaJes, 2017-07-17:
Hi 
I have the paypal account, but my country is locked for recieving funds...
I'm looking for another way to do it, asking my friends now

akaJes, 2017-07-25:
@ravini-mephisto i have made the instructable also https://www.instructables.com/id/Marlin-config/
maybe you have some suggestions how to improve it or my wrong language ? i hadn't an english classes in school :)

ravini-mephisto, 2017-07-26:
@akaJes  It is well done :)  Everything is easy to understand.

akaJes, 2017-08-17:
i made it :)
[image]

akaJes, 2017-08-20:
Thank you @ravini-mephisto for the donation

cleanbean, 2017-09-06:
@akaJes Thanks! awesome work, going to try to install on my soon to be octopi setup, it did work fine when i tried thru Qemu rasberry emulator, just need to verify that the serial communication works. when my pi has arrived

cleanbean, 2017-09-10:
Works like a charm on octopi, just update node to 4+ and install pip.
Donatio done!

akaJes, 2017-09-13:
Thank for donation!
I bought an 'orange pi pc plus'(ram 1gb flash 8gb wifi ethernet hdmi 3usb) and installed an armbian ubuntu server. Will try to setup my app too soon