akaJes/marlin-config #8: Custom boards?

gcormier, 2017-07-24:
What is the best way to add a custom board definition?

akaJes, 2017-07-25:
there is ability only for listed boards(supported with firmware) and also for the example configurations
the best way to support a custom board is to suggest it for marlin firmware community

gcormier, 2017-07-25:
It would be useful to be able to customize the pins somehow in this tool. The pins often change between various machines as a matter of customization (ie. bed sensor types using different pins). The Marlin FW doesn't want header files for every variation out there, but we still need a way to set these values. A potential future feature?

ravini-mephisto, 2017-07-26:
If you can follow the PlatformIO documentation (http://docs.platformio.org/en/latest/projectconf.html)
you will be able to customize the platformio.ini file that comes with Marlin.  
For example, take a look at the custom platformio.ini made my the Skynet3D project for Anet printer users: https://github.com/SkyNet3D/Marlin/blob/SkyNet3D-Devel/platformio.ini

akaJes, 2017-07-26:
@gcormier i looked at files with pin assignments - these are so various and have no common standart 
this idea is interesting but at this moment i have no plan how to implement it

closed 2020-03-04 by gcormier