akaJes/marlin-config #10: [Linux] desktop file is lacking a Categories= entry

probonopd, 2017-08-18:
The desktop file is lacking a `Categories=` entry in the `*.desktop` file.

According to the [menu spec](https://standards.freedesktop.org/menu-spec/latest/apa.html),

> By including one of the Main Categories in an application's desktop entry file, the application will be ensured that it will show up in a section of the application menu dedicated to this category. If multiple Main Categories are included in a single desktop entry file, the entry may appear more than once in the menu. 

Hence, please add at least one of the following in the `Categories=` key.

Main Category | Description | Notes
-- | -- | --
AudioVideo | Application for presenting, creating, or processing multimedia (audio/video) |  
Audio | An audio application | Desktop entry must include AudioVideo as well
Video | A video application | Desktop entry must include AudioVideo as well
Development | An application for development |  
Education | Educational software |  
Game | A game |  
Graphics | Application for viewing, creating, or processing graphics |  
Network | Network application such as a web browser |  
Office | An office type application |  
Science | Scientific software |  
Settings | Settings applications | Entries may appear in a separate menu or as part of a 		"Control Center"
System | System application, "System Tools" such as say a log viewer or network monitor |  
Utility | Small utility application, "Accessories"

In addition, you could specify one or more from the longer list of [Additional Categories](https://standards.freedesktop.org/menu-spec/latest/apas02.html).

Please test the result with `desktop-file-validate` and make sure it passes.

electron-builder offers native support for this since [v19.22.1](https://github.com/electron-userland/electron-builder/releases/tag/v19.22.1).

Reference: https://github.com/AppImage/AppImageHub/issues/2

akaJes, 2017-08-20:
Added in the 2.7.11

probonopd, 2017-08-20:
Thank you, added to https://github.com/AppImage/AppImageHub/commit/40e4ecac314d6035dfd3b9c2fc54cbd5e5a05779

closed 2017-08-20 by probonopd