akaJes/marlin-config #13: No filetabs for Configuration.h or Configuration_adv.h

ScruffR, 2017-10-13:
I'm running `mct` on Windows 8.1Pro 64bit but when I drag'n'drop or upload a file, I never see the respective file tabs appear in my browsers (Chrome 61.0.3163.100, FireFox  56.0.1 - IE 11 doesn't do anything at all - as expected :wink: )

This is the log in Chrome
```
jquery.min.js:4 [Deprecation] Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check https://xhr.spec.whatwg.org/.
send @ jquery.min.js:4
:3000/bs/custom Failed to load resource: the server responded with a status of 403 (Forbidden)
(unknown) Not found
check Failed to load resource: the server responded with a status of 403 (Forbidden)
```

Firewall is set up to allow access.

ScruffR, 2017-10-13:
I also keep getting this
```
fatal: Path 'Marlin\Configuration.h' exists on disk, but not in 'tags/1.1.6'.
(node:7468) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 7): fatal: Path 'Marlin\Configuration.h' exists on disk, but not in 'tags/1.1.6'.
fatal: Path 'Marlin\Configuration_adv.h' exists on disk, but not in 'tags/1.1.6'.
```

akaJes, 2017-10-13:
type in repository folder `git checkout 1.1.6 -f` - it resets to version 1.1.6 
maybe some issue with switching between branches

ScruffR, 2017-10-13:
No change
```
C:\MarlinConfTool\Marlin>dir
 Datenträger in Laufwerk C: ist Windows8_OS
 Volumeseriennummer: 4842-7795

 Verzeichnis von C:\MarlinConfTool\Marlin

13.10.2017  11:22    <DIR>          .
13.10.2017  11:22    <DIR>          ..
13.10.2017  11:18               367 .gitattributes
13.10.2017  11:21    <DIR>          .github
13.10.2017  11:21             1 968 .gitignore
13.10.2017  11:21            11 926 .travis.yml
13.10.2017  11:18    <DIR>          buildroot
13.10.2017  11:18            35 821 LICENSE
13.10.2017  11:22    <DIR>          Marlin
13.10.2017  11:22             1 922 platformio.ini
13.10.2017  11:22             6 576 README.md
               6 Datei(en),         58 580 Bytes
               5 Verzeichnis(se), 12 174 991 360 Bytes frei

C:\MarlinConfTool\Marlin>git checkout 1.1.6 -f
HEAD is now at f6f881f... Version 1.1.6

C:\MarlinConfTool\Marlin>cd ..

C:\MarlinConfTool>mct
Node v6.11.4
fatal: Not a git repository (or any of the parent directories): .git

no root
press Enter to clone it from web or ^C to exit
fatal: destination path 'Marlin' already exists and is not an empty directory.

[gitRoot] C:/MarlinConfTool/Marlin
loaded configuration hints
loaded gcodes
Marlin cam started on https://localhost:3002/
Marlin config tool started on http://localhost:3000/
SSE conected
list sent
[gitTag] tags/1.1.6
fatal: Path 'Marlin\Configuration.h' exists on disk, but not in 'tags/1.1.6'.

fatal: Path 'Marlin\Configuration_adv.h' exists on disk, but not in 'tags/1.1.6'.

fatal: Path 'Marlin\Configuration.h' exists on disk, but not in 'tags/1.1.6'.

(node:9196) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 2): fatal: Path 'Marlin\Configuration.h' exists on disk, but not in 'tags/1.1.6'.

SSE conected
list sent
fatal: Path 'Marlin\Configuration_adv.h' exists on disk, but not in 'tags/1.1.6'.

SSE removed
```

akaJes, 2017-10-13:
try to update to version 2.7.15 `npm -g i marlin-conf`

ScruffR, 2017-10-14:
Better! Now it works for me too 👍 

Thanks!

closed 2017-10-14 by ScruffR