akaJes/marlin-config #16: Compilation errors

witorwitor, 2017-11-08:
When I try to compile I get lots of errors. I have installed Python2.7.14. I have already tried 
pip install -U platformio 
and
pip install -U https://github.com/platformio/platformio-core/archive/develop.zip

Both seemed to work, but I am just getting this:

[11/08/17 00:05:15] Processing anet10 (platform: atmelavr; lib_deps: U8glib@1.19.1; upload_speed: 57600; board: sanguino_atmega1284p; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
Converting Marlin.ino
'avr-g++' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
AssertionError: :
File "C:\python27\lib\site-packages\platformio\builder\main.py", line 156:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Witor\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Witor\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "C:\Users\Witor\.platformio\platforms\atmelavr\builder\main.py", line 218:
target_elf = env.BuildProgram()
File "C:\Users\Witor\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 58:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Witor\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\platformio.py", line 256:
env.ConvertInoToCpp()
File "C:\Users\Witor\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\python27\lib\site-packages\platformio\builder\tools\piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "C:\python27\lib\site-packages\platformio\builder\tools\piomisc.py", line 54:
return self.process(contents)
File "C:\python27\lib\site-packages\platformio\builder\tools\piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
 [ERROR] Took 0.87 seconds 

 [SUMMARY] 
Environment megaatmega2560     	[SKIP]
Environment megaatmega1280     	[SKIP]
Environment printrboard        	[SKIP]
Environment anet10             	[ERROR]
 [ERROR] Took 0.87 seconds 
Environment printrboard_revf   	[SKIP]
Environment brainwavepro       	[SKIP]
Environment rambo              	[SKIP]
Environment sanguino_atmega644p	[SKIP]

Done.

akaJes, 2017-11-08:
Did you try to execute the command `platformio run` from terminal where placed `platformio.ini` directly?

witorwitor, 2017-11-08:
Yes and I get exactly the same list of errors

akaJes, 2017-11-09:
This is PlatformIO issue, as you understand. I can suggest only to remove
cache folders aka `.pio` from marlin project folder - it looks like the
toolchains dowloading errors