akaJes/marlin-config #17: Temperature - TEMP_SENSOR_BED (Bed Temperature) Missing Sensor

Chazeo, 2017-11-18:
TEMP_SENSOR_BED has no option to select Marlin Temperature Sensor of type, 75 : 100k Generic Silicon Heat Pad with NTC 100K MGB18-104F39050L32 thermistor.

akaJes, 2017-11-21:
it depends to source code of each version
if it exists in this line https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Configuration.h#L284
it will appears in selection box

another way is parse files like
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/thermistortables.h
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/thermistornames.h
but i have no confidence of its stable format

akaJes, 2017-12-11:
fixed in https://github.com/akaJes/marlin-config/releases/tag/v2.8.1

closed 2017-12-11 by akaJes