akaJes/marlin-config #18: Compile error after changing AXIS_RELATIVE_MODES in Configuration_adv

mvturnho, 2017-11-19:
After changing the setting AXIS_RELATIVE_MODES in Configuration_adv to True I get te compile error:

```
from Marlin\Marlin_main.cpp:244:
Marlin\Configuration_adv.h:360:29: error: initializer fails to determine size of 'axis_relative_modes'
#define AXIS_RELATIVE_MODES true
```

It seems Marlin-conf is not aware of the array of values and saves the new setting as:

`#define AXIS_RELATIVE_MODES true`

Where is should be someting like;

`#define AXIS_RELATIVE_MODES {true, true, true, true}`

akaJes, 2017-11-21:
yes i agreed with this,
i have ugly style of type definition https://github.com/akaJes/marlin-config/blob/master/app/mc-tool.js#L187

akaJes, 2017-12-11:
fixed in https://github.com/akaJes/marlin-config/releases/tag/v2.8.1

closed 2017-12-11 by akaJes