akaJes/marlin-config #21: Application crash MacOS (Marlin-conf 2.8.1 )

larsnordstrom, 2017-12-18:
Process:               marlin-conf [8244]
Path:                  /Applications/marlin-conf.app/Contents/MacOS/marlin-conf
Identifier:            com.akajes.marlin.conf
Version:               2.8.1 (2.8.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           marlin-conf [8244]
User ID:               501

Date/Time:             2017-12-18 18:22:43.692 +0100
OS Version:            Mac OS X 10.13.2 (17C88)
Report Version:        12
Bridge OS Version:     3.0 (14Y661)
Anonymous UUID:        505C0BB9-DFD7-155B-ECF3-BABC69BDD945

Sleep/Wake UUID:       1888F9CD-4197-4304-A181-93AAF9BD6787

Time Awake Since Boot: 270000 seconds
Time Since Wake:       13000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/Electron Framework.framework/Electron Framework
  Referenced from: /Applications/marlin-conf.app/Contents/MacOS/marlin-conf
  Reason: no suitable image found.  Did find:
	/Applications/marlin-conf.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: file too short
	/Applications/marlin-conf.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: stat() failed with errno=1

Binary Images:
       0x10be3b000 -        0x10be3bff7 +com.akajes.marlin.conf (2.8.1 - 2.8.1) <CB63147A-C9DC-308B-8CA1-EE92A5042E8E> /Applications/marlin-conf.app/Contents/MacOS/marlin-conf
       0x118c32000 -        0x118c7c98f  dyld (519.2.2) <6695F30B-4E88-3C0B-9867-7D738C44A3E6> /usr/lib/dyld

akaJes, 2017-12-19:
It was at start of application or after some operation?
Did you try previous versions?

rmeijsSdu, 2017-12-19:
Hi, I had a similar error on my Mac. Tried 2.8.0 which worked fine. My suggestion would be to have a look at the Electron Framework which the App apparently misses. Thanks for your work!

akaJes, 2017-12-20:
2.8.0 version was assembled with 1.7.8 Electron
2.8.1 with latest 1.7.9 
i have built with electron 1.7.10 for mac in https://github.com/akaJes/marlin-config/releases/tag/v2.8.1

larsnordstrom, 2017-12-20:
It was at application start when it crashed, tested the new version with electron 1.7.10 and it seem to work now.