akaJes/marlin-config #27: Support Marlin 32 bit firmware aka marlin 2.0.x

alfredanil, 2018-01-12:
Do you plan to support Marlin 2.0.x as well. If yes, when can it be expected?

akaJes, 2018-01-12:
it does
Versions menu -> branches
Compile -> (select 32bit platform)

alfredanil, 2018-01-18:
Error message:  
error: Your local changes to the following files would be overwritten by checkout: Marlin/Configuration.h Please commit your changes or stash them before you switch branches. Aborting
what do I do?

akaJes, 2018-01-22:
save your configuration with Zip, save or manual coping of configuration files 
use reset function
after changing version - just upload saved configuration files

Wheelycool, 2018-03-17:
I'm having the same issue. I need to use firmware for Melzi AtMega1284P. I can't get around the error message stated above. I manually copied the files. Reset => Branches => Latest version. Still get the message.

ModMike, 2018-05-07:
Having same issue, would like some clarity here as 1.18 is n longer an option.  I also want to use 2.0 for 8 bit board.

Any comments AKA Jes?