akaJes/marlin-config #31: Z_PROBE_ALLEN_KEY not configurable in Marlin config

kevinmce, 2018-02-12:
Hoped to use marlin config to maintain my machine byt as i now use Z_PROBE_ALLEN_KEY for a mechanical retracted Z probe i notice that this is ignored by marlin config. 

Hope to see support added, examples of this can be found in marlin example configurations -> Delta -> Kossel mini

akaJes, 2018-02-12:
support of all these options depend to a choosen marlin version....

[image]

akaJes, 2018-02-12:
start the application with empty folder and select menu items:
menu -> examples -> desired configuration (this only show changed options)
menu -> reset (here the main configuration will be overload with choosen from examples)

if options do not apear after restart the application(because at start the base configuration loaded from GIT), just select the same examples configuration, also you can see difference between the example and your main configurations

kevinmce, 2018-02-12:
I tried this but just without the reset. 

There was still a problem that the example configuration that held the Z_probe_allen_key was a Delta config and thus was missing quite a bit of settings for the cartesian printer config.  

Maybe the reset would solve the problem? 
As a workaround i just used marlin config to generate the latest bugfix branch of marlin, uploaded my older configuration.h & adv, then using arduino ide to paste in the section for the allen key probe before compiling.

akaJes, 2018-02-15:
I suggest to start use [PlatformIO](http://docs.platformio.org/en/latest/core.html) - it's hard to install on windows, but very helpful to use it 
auto loading for any platform:
-libraries
-compillers
-programmers
and uploading to board