akaJes/marlin-config #32: javascript errors in marlin-conf

windypop, 2018-02-15:
Trying to load Marlin on Anet A8, using marlin-config 2.8.1 both on Linux (Mint) and Win7 Pro I get the following report, and I can get no further... is there a work-around to correct the files?

**Uncaught Exception:
TypeError: data.NewProtocol.toLowerCase is not a function
    at /opt/marlin-conf/resources/app.asar/node_modules/nat-upnp/lib/nat-upnp/client.js:119:38
    at /opt/marlin-conf/resources/app.asar/node_modules/nat-upnp/lib/nat-upnp/device.js:158:9
    at Parser.<anonymous> (/opt/marlin-conf/resources/app.asar/node_modules/xml2js/lib/xml2js.js:199:18)
    at emitOne (events.js:96:13)
    at Parser.emit (events.js:191:7)
    at SAXParser.saxParser.onclosetag (/opt/marlin-conf/resources/app.asar/node_modules/xml2js/lib/xml2js.js:183:24)
    at emit (/opt/marlin-conf/resources/app.asar/node_modules/sax/lib/sax.js:624:35)
    at emitNode (/opt/marlin-conf/resources/app.asar/node_modules/sax/lib/sax.js:629:5)
    at closeTag (/opt/marlin-conf/resources/app.asar/node_modules/sax/lib/sax.js:889:7)
    at SAXParser.write (/opt/marlin-conf/resources/app.asar/node_modules/sax/lib/sax.js:1436:13)**

akaJes, 2018-02-15:
this issue invoked in third party library
i'll try to wrap this  code with the catch :)

windypop, 2018-02-15:
Thanks man.

Best regards

Mick Manning
01352781615
*mick@windyridge.eu <mick@windyridge.eu>*

akaJes, 2018-02-18:
i've generated this issue artificially and can't catch it my application
at this point you can reboot your router - it could help or look into uPnP settings of your router(maybe turn it off or reset)
i've created PR for this library and hope it will be fixed

windypop, 2018-02-19:
ok Jes, thanks for your help, I've just tried it on another linux box and a
win10 machine, the router reboots every night and I don't think it'll be
sensitive to data content.. could be the Java version but not as I see
it... is there an earlier version available that I could use, or do I just
forget the whole thing?

Best regards

Mick Manning
01352781615
*mick@windyridge.eu <mick@windyridge.eu>*

windypop, 2018-02-19:
.. and, sorry, it gives the same result with all 4 machines... not the
router either... any suggestions would be welcome though, I still wanna use
Marlin..

[image: Inline images 1]

Best regards

Mick Manning
01352781615
*mick@windyridge.eu <mick@windyridge.eu>*

akaJes, 2018-02-25:
checkout the new release for this issue

windypop, 2018-02-27:
Hi, thanks for that, tried it on Win7pro and LinuxMint18.2 - sadly no
change (other than the line numbers on the error message).

uPnP and NAT-PMP are both enabled on the router (Gargoyle on a
TPlink1034nd) but it's not showing any active port forwards when its
populating the folder (I assume it's working OK).

I've tried other routers - a stock firmware Cisco E4200 and a different
TPlink model with a different Gargoyle fw - all the same result! I'll
continue testing with other equipment and let you know if I find anything -
this is mysterious, it obviously works for other people!

Best regards

Mick Manning
01352781615
*mick@windyridge.eu <mick@windyridge.eu>*

akaJes, 2018-02-28:
Can you post the error lines, please

windypop, 2018-03-01:
[image]
in case the image is not readable, this is the text of the error message
_Uncaught Exception:
TypeError: data.NewProtocol.toLowerCase is not a function
    at /tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/nat-upnp/lib/nat-upnp/client.js:120:38
    at /tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/nat-upnp/lib/nat-upnp/device.js:158:9
    at Parser.<anonymous> (/tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/xml2js/lib/xml2js.js:199:18)
    at emitOne (events.js:115:13)
    at Parser.emit (events.js:210:7)
    at SAXParser.saxParser.onclosetag (/tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/xml2js/lib/xml2js.js:183:24)
    at emit (/tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/sax/lib/sax.js:624:35)
    at emitNode (/tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/sax/lib/sax.js:629:5)
    at closeTag (/tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/sax/lib/sax.js:889:7)
    at SAXParser.write (/tmp/.mount_marlinoZxs6Q/app/resources/app.asar/node_modules/sax/lib/sax.js:1436:13)_

akaJes, 2018-03-01:
try this one https://github.com/akaJes/marlin-config/releases/tag/v2.8.2-nat.2

windypop, 2018-03-02:
I know you're trying, but unsuccessful again I'm afraid - same error message (win7 and linux)!
Are there any ports I should open up, or other things I can do temporarily while it loads the files?

akaJes, 2018-03-04:
every your's error dump can help to solve this, because i can't reproduce your network state myself
could you try to start the application with disabled network/internet connection too?

windypop, 2018-03-06:
yeah that works fine thanks, just keeping internet access not available allows it to compile with no errors!!! - I don't understand, unless it has already downloaded all the stuff and redoing it upsets it somehow... So, issue resolved thanks again.
Anyway, now I have a different problem, the USB port isn't found on Win7 and is not allowed on Linux Mint (permissions!)...

closed 2018-03-06 by windypop