akaJes/marlin-config #33: Compiling Errors

ModMike, 2018-02-20:
I have flagged repeatedly and ran into compilation errors:

[image]

[image]

Can you help me out?

Thanks!

ModMike, 2018-02-20:
I managed to reinstall and compile again but still have the first error:

[image]

akaJes, 2018-02-23:
You colud remove the `.pioenvs` folder it will be recreated by PlatformIO
This looks like You run pio under administrative rights and than apllication under user rights

ModMike, 2018-02-23:
I can’t seem
To find the folder, even with a search.  Do you know where it is located on a Mac?

Thanks!

MT

akaJes, 2018-02-25:
Just deploy the firmware to new folder
or
move the visible folders to another new empty folder

ModMike, 2018-04-03:
I fixed the issue.  There is an error in the first section of the platformio.ini, which is in the top level directory.

This si how it is:

src_dir = Marlin
env_dir = .pioenvs
lib_dir = .piolib
libdeps_dir = .piolibdeps
env_default = megaatmega2560

This is how it should be:

[platformio]
src_dir = Marlin
**build_dir** = .pioenvs
lib_dir = .piolib
libdeps_dir = .piolibdeps
env_default = megaatmega2560

Changed the bold area and the error disappears.

closed 2018-04-03 by ModMike