akaJes/marlin-config #34: Can't run on macOS High Sierra (OS X 10.13.3)

DuhhBroken, 2018-02-27:
Hi,

Getting this error when running marlin-config on my mac.
Just built an ANET A8 printer and was told to start here first.
Have not connected the printer to the mac or installed drivers.
Advise if those are needed first.

Thanks!
[image]

Process:               marlin-conf [88582]
Path:                  /private/var/folders/*/marlin-conf.app/Contents/MacOS/marlin-conf
Identifier:            com.akajes.marlin.conf
Version:               2.8.2 (2.8.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           marlin-conf [88582]
User ID:               501

Date/Time:             2018-02-26 16:51:34.235 -0800
OS Version:            Mac OS X 10.13.3 (17D47)
Report Version:        12
Anonymous UUID:        32B292E4-E7BC-2E64-134C-58ACBE83533F

Sleep/Wake UUID:       EDDD8B4E-F8B5-497B-8B38-30C055372885

Time Awake Since Boot: 1100000 seconds
Time Since Wake:       3400 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/Electron Framework.framework/Electron Framework
  Referenced from: /private/var/folders/*/marlin-conf.app/Contents/MacOS/marlin-conf
  Reason: no suitable image found.  Did find:
	/private/var/folders/br/26g_c02s6s56hlryb5650kww0000gn/T/AppTranslocation/76ABE0FE-0252-43FD-892C-448AA75957DD/d/marlin-conf.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: file too short
	/private/var/folders/br/26g_c02s6s56hlryb5650kww0000gn/T/AppTranslocation/76ABE0FE-0252-43FD-892C-448AA75957DD/d/marlin-conf.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: stat() failed with errno=1

Binary Images:
       0x10a45e000 -        0x10a45eff7 +com.akajes.marlin.conf (2.8.2 - 2.8.2) <C11B8E11-EDD2-37B9-9AB5-5FE9E97FFFFB> /var/folders/*/marlin-conf.app/Contents/MacOS/marlin-conf
       0x10e02d000 -        0x10e07798f  dyld (519.2.2) <6695F30B-4E88-3C0B-9867-7D738C44A3E6> /usr/lib/dyld

Model: MacBookPro8,2, BootROM MBP81.004D.B00, 4 processors, Intel Core i7, 2.2 GHz, 16 GB, SMC 1.69f4
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Graphics: AMD Radeon HD 6750M, AMD Radeon HD 6750M, PCIe, 1 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D5341384758334D314131333333433920
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D5341384758334D314131333333433920
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xF5), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 6.0.2f2, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 850 PRO 512GB, 512.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

akaJes, 2018-02-28:
Thank you, did the previous version work?

DuhhBroken, 2018-02-28:
Have not run any other versions.
Is that what you would recommend?
If so, please provide a link to the version you would like me to try.

Thanks

akaJes, 2018-03-01:
https://github.com/akaJes/marlin-config/releases/download/v2.8.1/marlin-conf-2.8.1-mac-electron-1.7.10.tar.gz

thinkyhead, 2018-03-01:
That works, thanks!

akaJes, 2018-03-02:
@thinkyhead is the stuff really helps to speed up the work? maybe some suggestions for improvements?

cinealfa, 2018-03-02:
I'm having the same issue with  the 2.8.2

Process:               marlin-conf [41775]
Path:                  /private/var/folders/*/marlin-conf.app/Contents/MacOS/marlin-conf
Identifier:            com.akajes.marlin.conf
Version:               2.8.2 (2.8.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           marlin-conf [41775]
User ID:               502

Date/Time:             2018-03-02 03:14:21.901 -0800
OS Version:            Mac OS X 10.13.3 (17D47)
Report Version:        12
Anonymous UUID:        84533846-DDAC-D6B0-00F3-9199D7FD6E43

Sleep/Wake UUID:       8254916B-732F-4624-894B-2000E72D18FF

Time Awake Since Boot: 60000 seconds
Time Since Wake:       1000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/Electron Framework.framework/Electron Framework
  Referenced from: /private/var/folders/*/marlin-conf.app/Contents/MacOS/marlin-conf
  Reason: no suitable image found.  Did find:
	/private/var/folders/nt/c7fqldv531dcrh6nj_ljlgcw0000gp/T/AppTranslocation/A6D996F5-1B68-4BFA-886C-9896BF7F988B/d/marlin-conf.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: file too short
	/private/var/folders/nt/c7fqldv531dcrh6nj_ljlgcw0000gp/T/AppTranslocation/A6D996F5-1B68-4BFA-886C-9896BF7F988B/d/marlin-conf.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: stat() failed with errno=1

Binary Images:
       0x109a3a000 -        0x109a3aff7 +com.akajes.marlin.conf (2.8.2 - 2.8.2) <C11B8E11-EDD2-37B9-9AB5-5FE9E97FFFFB> /var/folders/*/marlin-conf.app/Contents/MacOS/marlin-conf
       0x116c1b000 -        0x116c6598f  dyld (519.2.2) <6695F30B-4E88-3C0B-9867-7D738C44A3E6> /usr/lib/dyld

Model: MacBookAir5,2, BootROM MBA51.00F4.B00, 2 processors, Intel Core i7, 2 GHz, 8 GB, SMC 2.5f9
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434353153364D465238412D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D5434353153364D465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xE9), Broadcom BCM43xx 1.0 (7.21.190.16.1a2)
Bluetooth: Version 6.0.2f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM512E, 500.28 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Air, Apple Inc., 23.4

DuhhBroken, 2018-03-02:
Tried the version you linked. Looks like the same error.

Process:               marlin-conf [99838]
Path:                  /private/var/folders/*/marlin-conf 2.app/Contents/MacOS/marlin-conf
Identifier:            com.akajes.marlin.conf
Version:               2.8.2 (2.8.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           marlin-conf [99838]
User ID:               501

Date/Time:             2018-03-02 06:40:00.036 -0800
OS Version:            Mac OS X 10.13.3 (17D47)
Report Version:        12
Anonymous UUID:        32B292E4-E7BC-2E64-134C-58ACBE83533F

Sleep/Wake UUID:       54A8E5A0-97B0-42FC-B632-094DA8873088

Time Awake Since Boot: 1200000 seconds
Time Since Wake:       120000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/Electron Framework.framework/Electron Framework
  Referenced from: /private/var/folders/*/marlin-conf 2.app/Contents/MacOS/marlin-conf
  Reason: no suitable image found.  Did find:
	/private/var/folders/br/26g_c02s6s56hlryb5650kww0000gn/T/AppTranslocation/8E879D00-01FC-492B-97E0-2F6382F2A3B7/d/marlin-conf 2.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: file too short
	/private/var/folders/br/26g_c02s6s56hlryb5650kww0000gn/T/AppTranslocation/8E879D00-01FC-492B-97E0-2F6382F2A3B7/d/marlin-conf 2.app/Contents/MacOS/../Frameworks/Electron Framework.framework/Electron Framework: stat() failed with errno=1

Binary Images:
       0x10bfe1000 -        0x10bfe1ff7 +com.akajes.marlin.conf (2.8.2 - 2.8.2) <C11B8E11-EDD2-37B9-9AB5-5FE9E97FFFFB> /var/folders/*/marlin-conf 2.app/Contents/MacOS/marlin-conf
       0x11a0f1000 -        0x11a13b98f  dyld (519.2.2) <6695F30B-4E88-3C0B-9867-7D738C44A3E6> /usr/lib/dyld

Model: MacBookPro8,2, BootROM MBP81.004D.B00, 4 processors, Intel Core i7, 2.2 GHz, 16 GB, SMC 1.69f4
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Graphics: AMD Radeon HD 6750M, AMD Radeon HD 6750M, PCIe, 1 GB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D5341384758334D314131333333433920
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D5341384758334D314131333333433920
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xF5), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 6.0.2f2, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 850 PRO 512GB, 512.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

cinealfa, 2018-03-02:
the previous version 2.8.1 never loads. it doesnt have any network traffic or error message

akaJes, 2018-03-02:
@DuhhBroken you posted the same 2.8.2 version error, but my link was for 2.8.1

thinkyhead, 2018-03-04:
> maybe some suggestions for improvements?

I'm digging it at the moment. I like having documentation alongside the options. The UI is a little sluggish. My main suggestion would be to hide things more until they're needed, perhaps by putting related options into separate tabbed panes.

DuhhBroken, 2018-03-08:
Sorry for the delay.
Had both versions of the Marlin app in the same folder.
Deleted both and re-downloaded the older version linked.

Works now! Thanks akaJes 👍

badde-media, 2018-03-16:
Versions 2.9 and 2.8.2 don't run under MacOS 10.12.6. The icon doesn't even appear, no process seems to be started.

akaJes, 2018-03-17:
it seems like only 1.7.10 version hasn't issues with mac