akaJes/marlin-config #35: mct git does not work on Windows

mcecchi, 2018-02-27:
mct git results in 'fatal: Path 'Marlin/Configuration_adv' does not exist in '1.1.4'' and 'fatal: Path 'Marlin/Configuration' does not exist in '1.1.4'' on Windows.

Console output:
```
C:\Users\mauro\Desktop\work\mct\ODM-1\Marlin>mct git 1.1.4 json
Node v8.9.4 marlin-conf v 2.8.2
[gitRoot] C:/Users/mauro/Desktop/work/mct/ODM-1/Marlin
fatal: Path 'Marlin/Configuration_adv' does not exist in '1.1.4'

(node:1208) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): fatal: Path 'Marlin/Configu
ration_adv' does not exist in '1.1.4'

(node:1208) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections
 that are not handled will terminate the Node.js process with a non-zero exit code.
[gitRoot] C:/Users/mauro/Desktop/work/mct/ODM-1/Marlin
fatal: Path 'Marlin/Configuration' does not exist in '1.1.4'

(node:1208) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 2): fatal: Path 'Marlin/Configu
ration' does not exist in '1.1.4'
```

akaJes, 2018-02-28:
i thought the old syle interface  nobody uses, i'll check out it soon

akaJes, 2018-03-17:
fixed https://github.com/akaJes/marlin-config/releases/tag/v2.9.1

closed 2018-03-20 by akaJes