akaJes/marlin-config #37: You shouldnt say PlatformIO is optional when it is compulsory

jimmybigbaws, 2018-03-04:
I cant compile anything as it says PlatformIO **HAS** to be installed. This wouldn't normally bug me but I'm no programmer and the hoops you have to jump through to try and install the PlatformIO core are beyond me. I already had PlatformIO IDE installed and have used it for other projects but apparently that doesn't suffice, from my perspective it seems like the core needs installed separately which i just cant manage

akaJes, 2018-03-04:
You can compile your firmware with Arduino or Atom

jimmybigbaws, 2018-03-04:
The program I have installed gives no options to use the Arduino IDE, even choosing boards from the drop sown does nothing. Every time i try to compile it says that atom/platformIO isn't installed.
I have both Arduino and Atom/platformIO installed prior to installing this,
I love the thought process behind it but am damned if i can get it to work in any sort of fashion

jimmybigbaws, 2018-03-04:
Don't know if this will post but ill try attaching a screenshot

[image]

akaJes, 2018-03-13:
this application doesn't replace the Arduino, just quick configure source files instant on your folder. You could open both and press Compile or Upload button in Andurino

akaJes, 2018-03-13:
or use PlatformIO Core if it installed

akaJes, 2018-03-17:
platformio installer embeded https://github.com/akaJes/marlin-config/releases/tag/v2.9.1