akaJes/marlin-config #39: Anet_10 for compiling not choosable?

DerDominik, 2018-03-17:
I used marlin-config several times, which works really good!
After update to v2.9 I cannot choose Anet_10 as target board in the compile option?!

Do I've to reinstall the target or do I've to choose another board?

akaJes, 2018-03-17:
look at  your marlin version selected, not all has ability to compile for your platform

akaJes, 2018-03-17:
it appears from 1.1.5 version

DerDominik, 2018-03-17:
Sorry, that I forget the most important information.

I tested first with marlin-1.1.8 - Everything ok and Anet_10 is choosable
After switching to actual marlin-bugfix-1.1.x - Anet_10 is missing

GhostlyCrowd, 2018-03-19:
This is not a bug in marlin-config, nor marlin. the Anet board is a Melzi clone so use sanguino_atmega1284p as its one and the same.

DerDominik, 2018-03-20:
Ah, ok.
Thanks for the information :-)
I'll try it later