akaJes/marlin-config #41: Melzi (Sanguino) atmega1284P not on list.

Wheelycool, 2018-03-18:
I believe this is why I keep getting "Error: The program size (69756 bytes) is greater than maximum allowed (64512 bytes)" during upload. The 1284P should have more space then 64512 bytes. The only choice on the compile list is 'sanguino_atmega644P'. What am I missing?

Wheelycool, 2018-03-18:
[image]

akaJes, 2018-03-19:
Some microchips has not enough memory space and for compiling need to remove the most memory consuming functions or abilities or even degrade version to most suitable

akaJes, 2018-03-19:
You could try to modify platformio.ini file by yourself for this processor according to the platformio documentation or google it

GhostlyCrowd, 2018-03-19:
the Anet10 is sanguino_atmega1284p as you see if you compare platformio.ini files from their other source trees.  

I dont know why they keep flp-flopping the name the end user sees in their source. but it is changed back to sanguino_atmega1284p in the 1.1.x bug fix and 2.x bugfix.

Wheelycool, 2018-03-26:
Finally got it to work, had to start with fresh code unmolested. Then update s/w, then load bugfix 2.0.X. Now compiling and uploading correctly. Thanks!

closed 2018-03-26 by Wheelycool