akaJes/marlin-config #43: Will not even begin compile

ModMike, 2018-03-31:
Fresh install on Windows 10.  Had to install python manually but everything else ok.  Running standard ramps 1.4 EFB, Com4, atmega 2560 clone.  Device shows up in devices as Atmega.  All good so far.

Copied Anet A8 configuration to config.h and config_adv.h t root, changed board to 2560.  Click compile and log says "Done" immediately.

What am I doing wrong?

akaJes, 2018-03-31:
Python is not the compiller, at start you could install PlatformIO compiller automatic with a one touch 
Than open empty folder wait a minute and drag your files in the application

akaJes, 2018-03-31:
At start choose desired version then drop the old configuration files in the application then push dropdown of compile button for board selection and compile it

ModMike, 2018-03-31:
I know python is not the compiler, I am just saying the install process did not work completely, I had to manually install Platform IO manually.  It would not install automatically no matter what I tried.

I don't understand what you mean by "drop the old configuration files".  Drag and drop the old files in or erase them?

akaJes, 2018-03-31:
Drag&drop or through menu 'upload-files' button, your files will be compared and loaded over existing settings, changed options will be highligted

closed 2018-04-12 by akaJes