akaJes/marlin-config #45: Feature Request: Ability to open an already downloaded local marlin folder

garylzimmer, 2018-04-11:
The default behavior of the program seems to expect me to start with a fresh marlin branch.
I would like to open a downloaded local marlin folder that has already been configured for further review and configuration using this program.

akaJes, 2018-04-12:
You can, but it must be a repository, the simplest is to overwrite your
folder over the fresh downloaded and then switched to the right version or
branch

Or the another way is create your own repository in your folder
```
git init
git add .
git commit -m'my version'
```
This would be a base version
Some exploring of the git documentation could get for you more information
about the tags and branches

pipersw, 2018-08-01:
I agree with boelle, I don't need the git software, you juste have to modify two files in marlin.
The must is a offline version

boelle, 2018-08-01:
i dont use the program anymore (and never did), way to much trouble. I spend less time editing the 2 config files, well the first time it took a bit more but now that i got the config right i just hold on to them and recycle them

BubbaJon, 2019-03-17:
I gotta say I'm a little more than annoyed at this "feature" right now.  The feature says "Open".  What it does is download a branch into the selected folder with absolutely no warning or asking permission to destroy anything there - which it just trashed a very complicated Marlin setup.  Please spare me the standard disclaimer about it being my fault I didn't back it up.  Software should do exactly what it says it's going to do and any deviation notify and ask permission to proceed.
If I coded like that I'd be fired in a flash.  How frickin hard is it to look first?