akaJes/marlin-config #46: Unable to flash firmware

dkircali, 2018-06-12:
Hi, before I was able to compile and flash the firmware using marlin-config. I have changed the PC, followed the same steps for installation, however I am not able to flash. I have tried both from web based interface and the normal one. The error I am getting:
```
ReferenceError: sore is not defined
    at C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\app\server.js:157:33
    at Layer.handle [as handle_request] (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\route.js:137:13)
    at Route.dispatch (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\route.js:112:3)
    at Layer.handle [as handle_request] (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\layer.js:95:5)
    at C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\index.js:281:22
    at param (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\index.js:354:14)
    at param (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\index.js:365:14)
    at param (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\index.js:365:14)
    at Function.process_params (C:\Users\dkirc\AppData\Roaming\npm\node_modules\marlin-conf\node_modules\express\lib\router\index.js:410:3)
```

What am I missing?

akaJes, 2018-06-13:
It's a typo https://github.com/akaJes/marlin-config/blob/master/app/server.js#L157 after refactoring
It will appears when you open before a console
Try to use older version before it will be fixed

Thanks

akaJes, 2018-06-13:
fixed in new version

closed 2018-06-13 by akaJes