akaJes/marlin-config #49: Delta configuration

focamonca, 2018-06-22:
Please, I need to configure a delta printer..
I can't set the delta option.. Can the developer help me?

akaJes, 2018-06-25:
You could select from the examples menu some of Delta printer
configuration, then use reset menu command and options must appear after it

focamonca, 2018-06-26:
I followed your instructions but I can't get for example "DELTA_DIAGONAL_ROD" or other delta parameters

htaxil, 2019-03-09:
Hi, I confirm that the options to configure a delta printer do not appear, even when they are present in the Configuration.h file edited with marlin-config tool.

#define DELTA
#define DELTA_SEGMENTS_PER_SECOND
#define DELTA_HOME_TO_SAFE_ZONE
#define DELTA_CALIBRATION_MENU
#define DELTA_AUTO_CALIBRATION
#define DELTA_CALIBRATION_DEFAULT_POINTS 
#define DELTA_CALIBRATION_RADIUS
#define PROBE_MANUALLY_STEP
#define DELTA_PRINTABLE_RADIUS
#define DELTA_DIAGONAL_ROD
#define DELTA_HEIGHT
#define DELTA_ENDSTOP_ADJ
#define DELTA_RADIUS
#define DELTA_TOWER_ANGLE_TRIM
#define DELTA_RADIUS_TRIM_TOWER
#define DELTA_DIAGONAL_ROD_TRIM_TOWER