akaJes/marlin-config #51: Install Error

ModMike, 2018-07-10:
I am trying to help someone install the configurator on a Windows 10 machine.  We used the panel to install Github and PIO but every time we go to restart, it says uninstalled.

Now we are getting:

javascript error has occurred in the main process uncaught exception: type header cannot read property headers

akaJes, 2018-07-17:
Which version installed and what kind of error? If possible leave some
screenshots

ModMike, 2018-07-17:
It was the latest version and I don't have screen shots.  The installer is completely broken, we had to install each piece manually.  You also say the PIO is optional when it clearly is NOT.  You need PIO to compile.

akaJes, 2018-07-17:
PIO is optional because the main feature of this app to configure headers
of your firmware. To compile the firmware is possible with your preferred
GUI like Arduino, atom or other. PIO is huge project and works better with
Linux like os and not included to this app. Also you can install it
manually from the official website

closed 2018-10-04 by akaJes