akaJes/marlin-config #52: problem compiling

jwt666, 2018-07-13:
Hi I am trying to put marlin on my anet a8. problem is when compiling I get the error message 
avr-gcc-ar: cannot find plugin 'liblto_plugin_0.dll'
any help would be great

choekben, 2018-07-13:
I'm am completely new to 3d printing, and a while back, my Anet A8 main board burnt out. I bought a MKS GEN V1.4. I tried compiling with the newest version of Marlin 1.1 bugfix, but this error message popped up:
Arduino: 1.8.5 (Mac OS X), Board: "Sanguino, ATmega644 or ATmega644A (16 MHz)"

sketch/temperature.cpp: In static member function 'static void Temperature::isr()':
temperature.cpp:2046: error: 'ADC_KEYPAD_PIN' was not declared in this scope
         START_ADC(ADC_KEYPAD_PIN);
                   ^
sketch/temperature.cpp:1954:55: note: in definition of macro 'SET_ADMUX_ADCSRA'
   #define SET_ADMUX_ADCSRA(pin) ADMUX = _BV(REFS0) | (pin & 0x07); SBI(ADCSRA, ADSC)
                                                       ^
sketch/temperature.cpp:2046:9: note: in expansion of macro 'START_ADC'
         START_ADC(ADC_KEYPAD_PIN);
         ^
exit status 1
'ADC_KEYPAD_PIN' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Here is my Configuration file: https://drive.google.com/file/d/1lll4m8PQ_CQRM3Z5SsSeLwa2g8MDWUgt/view?usp=sharing 
(link to google drive)
 Any help would be appreciated

akaJes, 2018-07-17:
@jwt666 it looks like some issue with compiller installation try to
reinstall Arduino/PlatformIO

akaJes, 2018-07-17:
@choekben you enabled `ADC_KEYPAD` option which works only with boards:
Ramps, anet 10 and sanguinololu 11

choekben, 2018-07-17:
@akaJes Thanks! how and where would I disable ADC_KEYPAD?

akaJes, 2018-07-17:
This option depends to `ZONESTAR_LCD `. If you really need this option, it
works only with listed above mainboards. Or just disable it