akaJes/marlin-config #55: 'avr-gcc' is not recognized?

hventaclu, 2018-07-27:
I know basically nothing about programming so I've just been following tutorials. The other night I went to compile settings and it worked fine but I didn't flash it because I knew I'd need to reverse my extruder motor after setting up the bowden setup i built tonight so I found the settings do so but when I went to compile it tonight I got these errors.

It made it through many lines until here.

Compiling .pioenvs\anet10\libe1a\U8glib_ID7\clib\u8g_bitmap.c.o
'avr-gcc' is not recognized as an internal or external command,
operable program or batch file.
'avr-gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pioenvs\anet10\libe1a\U8glib_ID7\clib\u8g_circle.c.o
*** [.pioenvs\anet10\libe1a\U8glib_ID7\clib\chessengine.c.o] Error 1
*** [.pioenvs\anet10\libe1a\U8glib_ID7\clib\u8g_bitmap.c.o] Error 1
'avr-gcc' is not recognized as an internal or external command,
operable program or batch file.
*** [.pioenvs\anet10\libe1a\U8glib_ID7\clib\u8g_circle.c.o] Error 1
 [ERROR] Took 11.12 seconds 

in the meantime I guess I'll just flip the plug.

akaJes, 2018-07-27:
You could remove the `.pioenv` folder from Marlin folder and it will be
downloaded again during next firmware compiling

hventaclu, 2018-07-28:
it's as above but it stops at chessengine.c.o

 Compiling .pioenvs\anet10\libe1a\U8glib_ID7\U8glib.cpp.o
Compiling .pioenvs\anet10\libe1a\U8glib_ID7\clib\chessengine.c.o
'avr-gcc' is not recognized as an internal or external command,
operable program or batch file.
*** [.pioenvs\anet10\libe1a\U8glib_ID7\clib\chessengine.c.o] Error 1

akaJes, 2018-07-30:
This is looks like the PlatformIO issue and needs to explore full logs, versions. Could you try to repeat all your steps using another computer?

hventaclu, 2018-07-31:
i'll give it a try. the antivirus software on this computer has been very aggressive in blocking things i download from github