akaJes/marlin-config #57: "this git not look like marlin repository"

GhostlyCrowd, 2018-08-07:
I went to change a setting on the same source i have been using for months and months, after updating Marlin-Config, and this is the error i get no matter what, even on a fresh git clone.

GhostlyCrowd, 2018-08-07:
I will add that, the git Desktop app sees this same repo fine, its just Marlin-Config.

akaJes, 2018-08-08:
Marlin config looks for the 'Marlin' folder in the root folder where the
git repository placed, maybe it renamed?

GhostlyCrowd, 2018-08-08:
Its not renames, I used marlin-config to clone a fresh repo, and at the end instead of opening it says the same error, but the repo clones fine.

yet-another-average-joe, 2018-10-02:
Did you use the same folder as root for your repo afeter reinstalling ?

After some manipulation, I had the same issue. It was just a hidden ".git" folder in my marlin-conf root directory. I deleted it, and it worked again, with no error message.

hierarchy :
d:\ marlin-conf \ Marlin \ repo with git and marlin files
the hidden .git folder was in d:\marlin-conf (it had nothing to do there !)

GhostlyCrowd, 2018-10-02:
No i tried even letting marlin-config clone and build the folder tree same issue, copied the same folder to a different computer marlin config works with it so i dont know i just gave up.

akaJes, 2018-10-03:
>  It was just a hidden ".git" folder in my marlin-conf root directory. I deleted it, and it worked again, with no error message.

The parentS folders could have this `.git` folder to invoke this issue

JCCasey, 2018-12-27:
> > It was just a hidden ".git" folder in my marlin-conf root directory. I deleted it, and it worked again, with no error message.
> 
> The parentS folders could have this `.git` folder to invoke this issue

Where do I find the marlin-conf root directory? I don't see any directory after installing from the exe file.