akaJes/marlin-config #58: Obsolete PIO Core v3.6.0 is used (previous was 3.6.1a4)

RudyBenner, 2018-09-15:
Obsolete PIO Core v3.6.0 is used (previous was 3.6.1a4)

Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system

The link is decidedly unhelpful.

Compiles fine with Arduino 1.8.7

RudyBenner, 2018-09-17:
I was hoping for a resolution. Noting has changed.

RudyBenner, 2018-09-17:
[image]
Deleted.

RudyBenner, 2018-09-17:
I will leave it open for a while.

akaJes, 2018-09-19:
Hi, which kind of error did you get and where?
The app detects if the command `pio` can be reached from a console and shows the installed state of PlatformIO.
Manual removing of this folder(on your image) uninstall the embedded PlatformIO of the app
 Then you can install your own version of PlatformIO manually

closed 2018-11-25 by RudyBenner