akaJes/marlin-config #60: [feature request] External access UPnP

yet-another-average-joe, 2018-09-23:
Hello akaJes !

marlin-conf, as far as I can understand, works as a web server. 

The External access UPnP is highly unsecure. https could be an improvement (personnaly, I think that a manual redirection is better, but this is out of topic...)

(sorry for my bad english)

akaJes, 2018-10-02:
Hi
For implemeting the https protocol everyone needs a certificate and domain - this is not easy configuration and depends to OS version
UPnP works only when tha app is started, another way could be used as simple authorization

yet-another-average-joe, 2018-10-02:
You did it for the camera preview, using an auto signed certificate

;)

akaJes, 2018-10-03:
Did only for browser restriction for video transfering
For local using only it allowed in local network with less of browser warnings
besides the certificate access doesn't guarantee more securing than any of kind an authorization