akaJes/marlin-config #61: [feature request] https warning / autosigned certificate

yet-another-average-joe, 2018-09-23:
Hello akaJes !

When using Remote camera preview snippet, browsers throw a warning about the autosigned certificate. Tested with Firefox and IE. This can be disturbing for the average joe.

It could be a good idea to put some explanation about what an autosigned SSL certificate is...

(sorry for my bad english)

akaJes, 2018-10-05:
can you show the explanation sample, which do you decide is better?

yet-another-average-joe, 2018-10-05:
When starting RaspBian/OctoPrint, just above the prompt, the system displays a list of active networks adapters and their IPs.

Just below this list, there is "https is also available, with a self-signed certificate"
It's very short !

So I think that near to the link to the camera display client and QR code, it would be fine to display something like :

"marlin-conf uses a self-signed SSL certificate, you will have to add an exception in your browser"

So the user understands why the browser is whiming : "connection is not safe", "<local IP> owner did not properly set the web site" and so on.

The way Remote camera display is presented is also somewhat disturbing. It shows an IP, and it is an hyperlink. Understanding that this link is to be opened on the phone, not on the machine that runs marlin-conf, is not obvious. (I had a hard time understanding that this IP opened ON THE PHONE redirects the video stream to it, and I imagine that other average joe like me will not understand immediatly)

Maybe, an explanation like "Activate WiFi and open this link on your phone or scan QR code" would be far better IMHO.

(remember that english is not my first langage !)

yet-another-average-joe, 2018-10-05:
Just something like this :

"marlin-conf uses an SSL self-signed certificate. You will have to add an exception in your browser"

Something is not obvious in the Remote camera preview. Maybe an explanation about how to use it :

"Activate WiFi on your phone, and open this link or scan this QR code"