akaJes/marlin-config #66: [feature request] Examples list alphabetic sorting

yet-another-average-joe, 2018-09-23:
Hello akaJes !

This is not an issue. But it would probably be if printers manufacturers gave their conf files ! When opening the examples list (from the main menu), the list is not sorted. Not a big issue, but a recursive alphabetic sort would be greet (Brand/Model)

Note that I don't use this feature as I don't need it. I'm just thinking of people who have listed printers like the popular Creality CR's for example.

akaJes, 2018-10-02:
This list is just folders names

yet-another-average-joe, 2018-10-02:
Yes indeed, but some sort would be great ! I did'nt look at the code, as I don't speek Java (C/C++ only for 30 years, and just a couple years for a living). I imagine this list is in an array, it's easy to implement a bubble sort on the array, and routines are certainly available, without having to code it by hand.

Examples have a weird behaviour on my system (Win7). When I select an example, the tabs contents is not updated, and config*.h files are not copied into Marlin folder. But the parameters are updated one by one when I click on the "reset parameter" button (the 'X'). The only way to play with examples is to copy files by hand, or use the upload feature. Tested on a real computer and a virtual machine I use as a marlin-conf server for testing purpose.

I will start tonight editiong a video with some other glitches, and I will let you informed (unreferenced video on Youtube, a link for your eyes only ; I already have some footage)

akaJes, 2018-10-03:
I can it sort at least
Choosing some from the Examples is just shows the Differences to your base configuration
The `reset` command do coping of files from examples or from git version

yet-another-average-joe, 2018-10-03:
On my system (win7), it does not copy example files. Will do a video soon (within 12-14 hours) in order to show you the weird behaviour with examples...

(yes, I still run win7 because one app I need does not run under 8 or 10)

akaJes, 2018-10-04:
fixed in [v2.10.5](https://github.com/akaJes/marlin-config/releases/tag/v2.10.5)

closed 2018-10-04 by akaJes