akaJes/marlin-config #68: [bug] RESET item in main menu

yet-another-average-joe, 2018-09-23:
Hello akaJes !

I found a problem with the RESET menu item. After some editing in configuration files, RESET seems not to work. In fact, it does...

First, open marlin-conf the normal way, selecting a repository from the main pannel. Modify one parameter, let's say SHOW_BOOTSCREEN. Turn if off. Configuration.h is immediatly modified, and it can be seen with a good editor as Notepad++, Visual Studio, etc. these editors sense file changes while file is opened, and ask for a reload...

Then select RESET. marlin-conf shows no change, but a warning "Oh smap {}". Nothing changes in marlin-conf, but the file itself is resetted OK.

Another test : do the same with a browser, opening the URL localhost:3000 (or whatever port it is if several instances of marlin-conf have been launched).

Same problem : same wrong "snap" message, display not updated, but file is resetted OK. Now, refresh the browser : the browser page now reflects the file reset.

It seems that something's wrong with this function. An irrelevant error message, an no display refresh.

For now, the fix I found is using marlin-conf in a browser, and use the refresh button after a reset.

As i wrote in a previous message, I am making tutorial vids about marlin-conf, and this appears to be really ennoying. I will wait until you answer before I continue because I wouldn't like presenting a workaround for this. (this being said, I'd like you to have a look to the future videos for approval,on a private channel before anything is published ! )

(sorry for my bad english)

yet-another-average-joe, 2018-10-02:
from Git GUI :

git-gui version : 0.21.GITGUI
git version 2.19.0.windows.1
git wrapper : C:/Program Files/Git/mingw64/libewec/git-core/git.exe
git exec dir : C:/Program Files/Git/mingw64/libewec/git-core/git-core
git-gui lib : C:/Program Files/Git/mingw64/share/git-gui/lib
Tcl/Tk 8.6.8

from command line, typing "git --version" :

git version 2.19.0.windows.1

I installed marlin-conf many times, from marlin-conf, as I want to test it in each and every aspect. I will try installing it standalone, before installing marlin-conf. Just in case. I wanted to test with Debian, but I failed (not really a Linux boy). Maybe, I will give it a try with Ubuntu. I feel like Windows/mingw could be a problem

(I have to say I understand (roughly) what Git does, but not how ! GUI is somewhat esoteric...)

akaJes, 2018-10-04:
fixed in [v2.10.5](https://github.com/akaJes/marlin-config/releases/tag/v2.10.5)

closed 2018-10-04 by akaJes