akaJes/marlin-config #69: [long term feature request] I had a dream

yet-another-average-joe, 2018-09-23:
Hello akaJes !

I had a dream, but I failed to make it real...

marlin-conf could be the killer app if it worked out of the box on an OctoPi print server.

I imagine a package for OctoPi, easy to install with just an "apt-get install marlin-conf", and running it as a service, and such a workflow :
- opening a SSL terminal from a computer
- killing the Octoprint service
- starting the marlin-conf server
- editing/upgrading the sources in a browser from a remote PC
- launching compile/flash
- killing marlin-conf server
- restarting Octoprint
- enjoy printing

Not being a Linux guru (by far !!!), I failed. An epic fail ! (I first failed at installing marlin-conf, because of dependencies I couldn't resolve)

IMHO, marlin-conf on a RasPi could make it even better. 100% independent from a desktop computer. The printer with a RasPi could be auto upgradable, even if the printer manufacturer does not maintain the firmware.

(sorry for my bad english)

akaJes, 2018-10-04:
it can be installed with `npm -g i marlin-conf` and run as service with `mct conf` #11

ScruffR, 2020-01-07:
Sorry for resurrecting this old issue, but I cannot.
For some reason the install always fails with something like this
```
pi@octopi:~ $ sudo npm -g i marlin-conf
npm WARN deprecated rtcmulticonnection-v3@3.4.5: Please use `rtcmulticonnection` instead. i.e. do NOT include `-v3`.
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/package.json'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/AUTHORS'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/CHANGELOG.md'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/index.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/request-08a12583/index.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/LICENSE.md'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/request-08a12583/LICENSE'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/README.md'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uglify-js-232219aa/tools/props.html'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/request-08a12583/README.md'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/v1.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/v3.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/v4.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/uuid-d98ec939/v5.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/node-notifier-29fd484b/vendor/snoreToast/SnoreToast.exe'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/bootstrap-308187b2/dist/css/bootstrap.min.css.map'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/font-awesome-64e5ea73/fonts/fontawesome-webfont.ttf'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/socket.io-client-e85870ce/dist/socket.io.slim.dev.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/sockjs-client-c86bdff1/dist/sockjs.js.map'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/handlebars-7f3b16c0/dist/amd/handlebars/compiler/printer.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/handlebars-7f3b16c0/dist/amd/handlebars/compiler/visitor.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/handlebars-7f3b16c0/dist/amd/handlebars/compiler/whitespace-control.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/handlebars-7f3b16c0/dist/amd/handlebars/decorators.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/ace-builds-8fedac08/src-min-noconflict/ext-beautify.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/ace-builds-8fedac08/src-min-noconflict/ext-elastic_tabstops_lite.js'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/ace-builds-8fedac08/src-min-noconflict/ext-emmet.js'
npm WARN tar ENOENT: no such file or directory, lstat '/usr/local/lib/node_modules/.staging/vex-js-a51109ac/docs/welcome'
npm WARN tar ENOENT: no such file or directory, open '/usr/local/lib/node_modules/.staging/moment-08b2c421/min/moment-with-locales.js'
npm WARN notsup Unsupported engine for rtcmulticonnection-v3@3.4.5: wanted: {"node":"latest"} (current: {"node":"12.14.0","npm":"6.13.4"})
npm WARN notsup Not compatible with your version of node/npm: rtcmulticonnection-v3@3.4.5

npm ERR! code 1
npm ERR! Command failed: git clone --mirror -q https://github.com/MelvinTo/node-nat-upnp.git /root/.npm/_cacache/tmp/git-clone-e127d5d1/.git
npm ERR!

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-07T13_04_50_749Z-debug.log
```

ScruffR, 2022-02-03:
Now one and a half years later with a fresh install, my previous solution is not working anymore
See #119