akaJes/marlin-config #77: [bug] editor / tab not synched

yet-another-average-joe, 2018-10-04:
Hello,

after having edited a config* file, the file is updated on disk (it seems to me it's done in real time)

but after closing it, or while editing, the data are not updated in the tab (for example Configuration while editing Configuration.h). marlin-conf has to be restarted in order to display the changes.

akaJes, 2018-10-05:
This editor intended for editing other files except Configuration*.h

yet-another-average-joe, 2018-10-10:
It could !

This just needs that some sort of reload command is issued after the editor is closed. But it would need the editor to be executed in a modal dialog/window (sorry, but I only speek "Windows API"...)

Or (maybe better...), a "reload" command in the main menu. This would allow the user to use an external editor (Notepad++, Visual Studio, etc.), and manually reload the files in marlin-config.

Or file changes detection ; not sure it's easy to code for a multioplatform app ; depends on file management and system notifications). A "Reload" menu item could be a great feature).

Or simply open this files in read only mode, with a warning message !