akaJes/marlin-config #78: 2.10.6 is broken !

yet-another-average-joe, 2018-10-05:
Hello,

I was very excited about 2.10.6, but :

- it takes 4 minutes to load (tested several times with Win7 and Win10) : blank marlin-conf window, blank browser page (using localhost:3000 or whatever). No disk activity, no network activity, no processor activity. Same 4 minutes when switching to another Marlin version, and also when hitting Reset.

- examples menu item won't open examples list

- no more Editor (the new one seems to be gorgeous !)

tested with Win7 and Win10, tested after uninstalling previous version and also installing over previous version, Win7 and Win10.

I had to revert to 2.10.5, as I am testing marlin-conf vs mct about firmware upgrades (some differences found)

GhostlyCrowd, 2018-10-05:
Can confirm same issues

RudyBenner, 2018-10-06:
Was running older version. Tried to update. No go.

RudyBenner, 2018-10-06:
Ok, version 2.10.3 works. I am back to where I was. The newer versions go off into the ionosphere.

RudyBenner, 2018-10-06:
Tried to update to 2.10.6. No luck at all. Tried older versions. 2.10.3 works.

akaJes, 2018-10-09:
Fixed in 2.10.7

closed 2018-10-09 by akaJes