akaJes/marlin-config #84: No pull synchro with the marlin repo

axl12ap, 2018-10-19:
Good morning,
with the latest version of marlin-conf 2.10.7 (npm package 2.10.5) I've a problem when I try to get the latest marlin repo version.

Opening marlin-conf, tell a new folder to get all sourcers from Git, waiting that program pull all sources.
Then I switch branches "origin/bugfix-2.0.x" and wait for the updates of source code (tried also to update with UPDATE button).

If then I compare the folder marlin-conf has downloaded and a ZIP zip version of latest marlin Git  repo, there're a lot of difference in files.

Please check it.

akaJes, 2018-10-19:
The app uses `git fetch` command and it updates only the repository not
working folder, to do it would use 'reset' menu command
https://i.stack.imgur.com/nWYnQ.png

axl12ap, 2018-10-19:
Sorry, I don't understood, please explain me the correct sequence to have the working "bugfix-2.0.x" folder on my local drive with marlin-conf

yet-another-average-joe, 2018-10-26:
Hello,

This is not an issue, it's a feature ! And this is not a forum, it is intended for bug reports and feature requests. But I can try to explain... (a forum dedicated to marlin-conf would be great !)

As I understand Marlin-Conf, it is not intended to be used the way you attempt to use it ! You cannot change version (clone repository) unless you cancel all changes. This is what akaJes explained : Reset command. Switching versions from the menu would probably be a mess ! Imagine switching from a newer Marlin version  to an older one : lots of #define would be lost !

So, if you want to keep your current version untouched :

Create a new folder for the latest Marlin 2.x bugfix. Then open this foder with Marlin-Conf. It will clone the last 1.1x repository. Then, change for branch 2.x. You will get a vanila 2.x (last bugfix)

Now, compare config*.h versions (open files and search for CONFIGURATION_H_VERSION and CONFIGURATION_ADV_H_VERSION). If versions are the same, you can copy old config*.h files to the newly created folder, overwriting the vanila ones. Don't forget other files if needed : _Bootscreen.h, _Statusscreen.h.

If version numbers differ, drag the files in marlin-conf window, and they will be upgraded. You can also do this if headers versions are the same, they will remain the same (to be verified, I did'nt !)

After you do a "drag/drop upgrade", it is a good idea to open 2 marlin-conf instances, one for the old Marlin version and one for the new one. Then, compare the values. Report any problem to akaJes.

You can also cancel all changes (backup your config*.h files !) : Reset. This is what akaJes explained to you ! Then restore your files (if versions are the same), or simply drag/drop. On my side, I prefer keeping multiple versions, each one in its own directory. It's easier to switch (and flash) from one version to another.

robustini, 2018-10-30:
When i try to change branch from 1.1.x to 2.0 bug-fix i've a lot of error, i don't want to commit anything, i'm not a Marlin developer, i want just switch to the other branch with my settings... this is not intuitive in your configurator.

[image]

robustini, 2018-10-30:
Where is the marlin-conf cfg file? I want to erase all the customized directory list...

yet-another-average-joe, 2018-10-30:
These files are in the vendor's source archive : 

Configuration.h
Configuration_adv.h

and 2 others, optionnal :

_Bootscreen.h
_Statusscreen.h

Be carefull. There's a known bug with some parameters when you use a BLTouch. (some printers come with a BLTouch). You will have to edit these parameters by hand after upgrade)

robustini, 2018-10-30:
I mean "how to erase the custom directory list" from the configurator...

[image]

akaJes, 2018-11-02:
`~/.data-store/marlin-config.json`
i belive that you know about the `~` is the user's home folder?

akaJes, 2018-11-02:
@robustini before version changing or updating changes from branch it must be resetted from RESET menu command

yet-another-average-joe, 2018-11-02:
On Windows OS, "~/.data-store" becomes "[system disk]\users\[current user]\data-store", but this is not really usefull, as the "new" button and the MRU list do nothing when clicking, Furthermore, the viewport is too small to display full paths. The "open folder" button remembers the last opened folder, and this is OK.

akaJes, 2018-11-05:
The MRU has popups for each row with full path, maybe a windows has some issue with displaying?

yet-another-average-joe, 2018-11-05:
Yes, as showned by the image posted by Robustini, the viewport is too small, and the paths are not cliclable (they do nothing).

This is not really an issue. It is just cosmetic.

Under Windows at least, the MRU is not that usefull. When clicking the button "open folder", it opens an OpenFileDialog pointing to the last opened firmware, and everything can be done from there : browsing, creating a new folder, renaming, etc. The user will probably create all his working folders as a subfolder of a root folder, and, after cloning, rename each and every firmware (the "Marlin" folder created by Git).