akaJes/marlin-config #90: OSX: blank screen

Vanuan, 2018-12-07:
When opening configuration folder I see blank screen

Vanuan, 2018-12-07:
Turns out the issue was that 3000 port was already used by another app.

So marlin-config should either add a warning about 3000 port being already in use or choose the port dynamically (and show it somewhere in UI if needed).

Why do you need to run webserver in the first place?

yet-another-average-joe, 2018-12-07:
I'm NOT a marlin-conf developper, but I have been testing it extensively.

Normally, when 3000 is in use, marlin-conf uses other ports : 50xxx
Maybe you could solve the problem opening a 2nd instance of marlin-conf, and then closing the first (blank an thus useless) one.

On the Git main page, you can read :

Main features :
...
collaborative editor for whole project
...

marlin-conf, as a web server, allows multiple users to work over LAN and WAN (using a port redirection). It is a collaborative tool. Open a source folder with 2 instances or more, locally or over a LAN/WAN network, and you will see that changes are synchronized. marlin-conf can also be installed on a remote machine. Note that I failed installing it on a Raspi along with OctoPrint, but I'm not fluent with RaspBian... I successfully got it to work on my personal (Windows) server, over LAN and WAN. The server/client model is great. marlin-conf runs in a (hidden) Chrome window. OctoPrint is also a web server...

Vanuan, 2018-12-08:
Ok, so the issue is not that marlin can't provide an error message when 3000 is in use. The issue is that 3000 being in use by some other app is expected behavior.

Hm... I find it hard to sound reasonable. If marlin-conf is really supposed to be 2 different applications: client and server, users should choose themselves whether they want embedded client from marlin-conf or open localhost:3000 in the browser. And server should write a message: "I'm not starting since port 3000 is already in use by another instance".

Vanuan, 2018-12-08:
Alternatively, user should be able to override the port.

Think about it in this way:
if every electron app used a static port, you would only be able to run one app on your computer. That's like returning to 90s.

yet-another-average-joe, 2018-12-08:
When I said client server, it was inappropriate. What I meant is that marlin-conf is a web server. and the clients the browser windows. (client is not correct). When using the standalone app, the window is in fact a minimal browser : a Chrome window embeded in the "client" app window.

[image]

What happens when 3000 is already in use (by marlin-conf itself, so this is a 2nd instance) : does not seem to be that static... But you are right, it would be fine to have the option of choosing the ports. I've been searching for ports in some configuration file, but I failed. And I can't read JS.

[image]