akaJes/marlin-config #95: Missing anet10 compiler

haleyJ121, 2019-02-19:
[image]

Not sure if this has been posted, but I created this account in hopes of solving this issue. I am trying to get marlin configurator to work for the first time (new to all this) and am running into possibly simple problems that are preventing me from loading marlin 1.1.9 onto my Anet A8. From what I have seen, there needs to be an anet10 compiler in this list, as to properly compile for my specific board (the stock A8 board) I don't know how to do that and would appreciate any and all help

yet-another-average-joe, 2019-02-22:
Hi

there's no specific compiler. The compiler is always the same : GCC : GNU C Compiler, that you will find in many IDE's (Integrated Development Environments), including marlin-config. What you see here is only "precompiler directives", that will say to the precompiler : "prepare code for this hardware". 

IMHO, begining with with marlin-config is not the best idea. You'd better try with Arduino IDE, and learn some basics with. Compiling and flashing with Arduino IDE, you will understand what marlin-config does. You will find a huge amount of tutorials on the web and on Youtube. After a few hours, go back to marlin-config, and it will become obvious to you. You don't need to speek C. You just need to understand what a precompiler+compiler do. 1 or 2 untertaining hours.

I cannot speek in place of akajes. I hope everything's fine for him. But we had no news for a while... As explained by the developper of Marlin (thinkyhead), akajes has no direct help nor colalboration for marlin-config development. You will find an interview with thinkyhead by Thomas SanLaderer (maybe august 2018). This is sad... marlin-config is great.

Radu10, 2019-04-24:
Hello All ,i just buy a Anet A6 , and just try to update the firmware with copile  ,,Sangino_atmega1281p'' and is not working .
 I will try now with Metzi,