akaJes/marlin-config #97: Windows 10 - Nothing happens after having selected the folder

kalareor, 2019-03-01:
Hi there !

I am desperately trying to get this thing to work on my personal Windows 10 machine. I followed the installation steps including installing GIT and platformIO. Marlin-conf 2.10.7 launches without problems, recognizes that dependencies are all installed and downloads the Marlin repo after I select my newly  created folder. Then the window with the the four parts (GIT (installed), platformIO (installed), Configurations and Select folder at 100%) closes and then... Well, basically nothing. No new window opening, no response from localhost:3000 or [my_local_ip]:3000. and neither Power Shell nor Command Prompt recognizes the "mct" command.

I am a bit lost here and some help would be very welcome. I checked as far as I could on internet for same issue but I didn't find any viable solution apart from running Marlin-conf on a Linux or Mac machine which I would like to avoid if possible. If I missed the answer to my question somewhere in another thread, I apologize in advance and would still gladly take the link ;)

Thank you very much in advance.

Kala

yet-another-average-joe, 2019-03-02:
Some details are missing :

- did you allow Windows Firewall to create an exception for marlin-conf.exe when it asked for ? (just after the repo was downloaded)  
- did you edit the PATH environment variable ? (needed for python)
- did you install node.js and "npm'ed" marlin-conf ? (needed for mct)

(localhost:3000 will work only if marlin-conf IDE is running)

kalareor, 2019-03-02:
Hi Joe,

Thanks for your answer.

1. Firewall: Yes
2. PATH environment table : No
3. node.js and "npm ed": No

For points 2. and 3. I have no clue what you are talking about/how to proceed as I cannot find any **_windows_** instruction for them (or at least instructions I understand). I basically followed the "If You open this page and don't understand how to install - just download binaries https://github.com/akaJes/marlin-config/releases", that's what I did, I just downloaded "marlin-conf-setup-2.10.7.exe" and followed what the program was asking me to do.

Again sorry for the bother and thanks in advance.

Cheers !

Kala

yet-another-average-joe, 2019-03-02:
You're welcome.

(for the firewall, it has to be for private networks)

There is an installation bug : you MUST add the path to Python, otherwise platformio won't work. But this is only a problem for compile/flash functions, not for the IDE. See here :

https://github.com/akaJes/marlin-config/issues/59

mct is a different animal : it relies on node.js. You have to install node.js first, and then mct. See here : 

https://www.npmjs.com/package/marlin-conf 

it is not installed by the marlin-conf installer but it is not required for marlin-conf to work !

(instructions are somewhat unclear, that's true !)

But after your working directory is created, you should have a window with the IDE !

Doing (never released) videos about marlin-conf,  I made maybe 20 installations on virtual machines (Win7 Pro, 8.1 Pro, 10 Pro and LTSB), and never faced such a problem (nothing happening when pushing  the open folder button...

This is weird. Could an antivirus be the culprit ? (I can't answer, I don't use AVs, including the integrated one I always deactivate, I just use the firewall in paranoid mode)

Just to see, maybe you could disconnect your computer from the Internet (deactivating the adapter or unpluging the cable), and temporarily deactivate Windows Firewall ? marlin-conf being a web server, if there is nothing when you open it, it could be an internal network problem.

kalareor, 2019-03-02:
Hi again,

And thanks for the detailed answer :)

I spent another couple of hours trying stuff around and here are some facts (including or not thing you mentioned above).

- I run my Windows 10 machine as Admin all the time, so there should not be any problems on this level.
- Triple checked my Firewall settings. Both Private and Public Networks are checked for Marlin-conf.
- Tried to disable Firewall, Anti-virus (Avast 19.2.2364 (build 19.2.4186.436)), both at the same time and even uninstall anti-virus... Same story.
- PlatfromIO binaries installation bug: I know it's sort of optional but as I intend to compile my firmware I figured this one out by installing PlatformIO through Visual Studio Code. As mentioned in my first post the Marlin-config window checking the dependencies considers it installed. Please find attached underneath a screenshot of this window right after the repo cloning and just before closing itself without any further activity.

[image]

- I additionally went through the detailed installation procedure on the "hints for Windows-installation" page (https://github.com/akaJes/marlin-config/wiki/Windows-Installation) and I am basically having the exact same problem... Here is an extract of my cmd window regarding all the checks and mct launching (I replaced the help output with [...] for easier reading : 

Microsoft Windows [Version 10.0.17134.286]
(c) 2018 Microsoft Corporation. All rights reserved.

D:\Desktop\Test>node --version
v10.15.2

D:\Desktop\Test>node --help
Usage: node [options] [ -e script | script.js | - ] [arguments]
       node inspect script.js [arguments]

[...]

D:\Desktop\Test>git --version
git version 2.21.0.windows.1

D:\Desktop\Test>git --help
usage: git [--version] [--help] [-C <path>] [-c <name>=<value>]
           [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
           [-p | --paginate | -P | --no-pager] [--no-replace-objects] [--bare]
           [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
           <command> [<args>]

[...]

D:\Desktop\Test>npm --version
6.4.1

D:\Desktop\Test>npm --help

Usage: npm <command>

[...]

D:\Desktop\Test>mct help
Node v10.15.2 marlin-conf v 2.10.5
marlin-conf v 2.10.5
usage: mct help|git|tree|conf|clone

[...]

D:\Desktop\Test>mct
Node v10.15.2 marlin-conf v 2.10.5
fatal: not a git repository (or any of the parent directories): .git

no root
press Enter to clone it from web or ^C to exit
......................
[gitRoot] D:/Desktop/Test/Marlin

D:\Desktop\Test>_

After that again, nothing happens. No GUI poping-out, no response from localhost:3000 (tried in Chrome, Firefox and Edge). The repo is cloned correctly though as I have a proper full Marlin folder in my Test folder.

- Tried to select a folder on my C: drive where my Windows machine is installed (I have C: for the system only and D: for everything else) as I had hard drive location problems before with other software. Not helping either.

As I mentioned before I have access at work to a Mac machine to try make it run there but as I intend to tweek the firmware for my personal 3D printer at home (Printrbot SImple Metal with PrintrBoard Rev. F6) it would be really convenient to have Marlin-config running on my personal Windows machine. I am also trying to use Arduino for compiling my firmware but without success so far as I need a combination of at90usb1286 and teensy2++ which is a bloody nightmare to get to work correctly on Arduino.

Well I am also new to all this so I am far from knowing all the perks of firmware compiling and this is why maybe I struggle so much.

Again, thanks a lot and if I can be of any help to fix this issue it would be with pleasure. Otherwise I will play Firmware ping-pong between my work and my home :P 

Cheers!

Kala

yet-another-average-joe, 2019-03-02:
Sorry, but I don't know how to help... Everything seems to be OK.

Could the port 3000 be used by something else ? A few month ago, someone reported an issue with that. 

You have at least one option : installing everything in a virtual machine (VirtualBox for example). It should work. I used while I was testing, before installing on the real machiens. Fresh OS, fresh marlin-conf. 2 hours.

You already had platformio installed before you installed marlin-conf ? (I don't know anything about platformio as I'm a Visual Studio and Visual Micro fanboy...). This could explain why you did'nt have the PATH issue. Google just told me that platformio default port is 8008, not 3000.

And no news from akajes...

kalareor, 2019-03-03:
Hi Joe!

No worries!

In the meantime I figured out how to compile my firmware with Arduino + Teensyduino and flash my 3d printer with FLIP, so I got my things going ;)

Regarding the ports... Funny thing is that before selecting the folder, marlin-config process (in the task manager) is active and actually opens 3000 and 3002 ports. However as soon after I select the folder (and the non appearance of the GUI) the marlin-config process goes to "Terminated" and disappears from the task manager after a minute or so...  No other process uses the 3000 or 3002 ports.

Regarding platformIO I tried to install it the first time through the Marlin-config "Auto Install" button, but after every time relaunching marlin-config it was still not recognized as installed. So I just went to official platformIO web-site and followed their installation procedure through Visual Studio Code. This fixed the auto-install problem. However I don't know if platformIO works as I never got to the GUI of Marlin-config to test it :P

It will certainly be a very stupid question which have a perfectly logical answer, but if Marlin-config have  already a web-interface through localhost:3000 why don't you turn it into a web-site based tool where anybody could upload their Configuration.h and Configuration_adv.h files and like in the current software choose their board, marlin version etc... and get out their compiled hex file ? (ok, not sure how difficult it will be to integrate the flashing part into a web-site based tool, but at least the compiling would be awesome).

I mean this would be an awesome web-tool for which I would gratefully make a donation to use without the installation hassle it is a bit suffering from now (at least in my case :P).

Anyway, thanks again for your help and sorry for the hassle. Good luck with you project and i will make sure check on it from time to time to see if any new versions would work on my machine or try again when I do a clean install on Windows 10.

Cheers!

Kala

closed 2019-03-04 by kalareor

yet-another-average-joe, 2019-03-04:
> marlin-config process (in the task manager) is active and actually opens 3000 and 3002 ports< 

3000 is for marlin-conf, 3002 is for a phone used as a webcam (or another port : 3003, and so on if already in use)

> every time relaunching marlin-config it was still not recognized as installed <

because the path to Python is not updated by the installer ; the problem is solved with the fix I gave here #59 

> I just went to official platformIO web-site and followed their installation procedure through Visual Studio Code

the problem is maybe there. This is why I asked about this.

(I am NOT the developper)