akaJes/marlin-config #103: Board

NA7KR, 2019-08-08:
How do I add a board not in your list? (FYSETC F6)

akaJes, 2019-08-21:
Hi, You need for this board in the file https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/boards.h:
* look for similar board from the list
* wait when it appears in new versions
* modify personal for you in folder manually
* create PR in the MarlinFirmware repo, when You have working version

jwidht, 2020-04-21:
If you look through this video, https://youtu.be/vCYUnPoZSj4?t=428, the author is able to select anet10 board from the list. However, it is not in the list now. My guess is that the combobox is missing a scrolbar, and there are too many boards, the boards at the bottom are impossible to select.
Is there a workaround other then using a previous version as suggested in the comments? Thanks!

peacekeeper03, 2020-12-23:
Same Problem. There are a massive list of boards but without the option to scroll down. If i pivot the monitor i see more but not the end of the list.
Is there a way to remove (disable) some boards to shorten the list?