akaJes/marlin-config #104: bugfix-2.0.x no longer loading as of commit 274934a

Pyro-Fox, 2019-08-17:
I tried to grab the latest Marlin bugfix-2.0.x branch using Marlin-Conf and the UI no longer loads up the config file. I kept stepping back through commits until it worked and can see that it broke after [274934a](https://github.com/MarlinFirmware/Marlin/commit/274934ad81ee6cd9b4f686e55107966cc4e23bd7).

I really don't understand what might be happening other than,
Marlin-Conf works on [179d6c4](https://github.com/MarlinFirmware/Marlin/commit/179d6c4ed115e67ae826b5738e0009c6be2d58ea) from 8/14/2019...
[image]

...but totally breaks on the next commit [274934a](https://github.com/MarlinFirmware/Marlin/commit/274934ad81ee6cd9b4f686e55107966cc4e23bd7) from 8/16/2019.
[image]

Pyro-Fox, 2019-08-17:
Related pull request from Marlin: https://github.com/MarlinFirmware/Marlin/pull/14967

akaJes, 2019-08-21:
Hi, the App catch a fail and stops when it was parsed this wrong text
```
//#defined LULZBOT_TOUCH_UI
```
this is a two sided issue
1. wrong comment in the source Configuration.h
2. prevent this situation and fix the regexp in the line
https://github.com/akaJes/marlin-config/blob/master/app/mc.js#L37

akaJes, 2019-08-21:
it was marked as typo and fixed in the source here https://github.com/MarlinFirmware/Marlin/blame/bugfix-2.0.x/Marlin/Configuration.h#L2041

yet-another-average-joe, 2019-08-25:
@akaJes : happy to read you again !