akaJes/marlin-config #105: remove folders history

madmax2, 2019-09-07:
how do I remove all the folders history of marlin that I have opened?
I would like to clean it up to remove older folders that I no longer use?

https://i.imgur.com/p5DcZh4.png

akaJes, 2019-09-07:
Hi, remove in home folder the '.data-store' directory, or harder way is to
edit marlin-config.json file inside it

madmax2, 2019-09-07:
where is the home folder that contains .data-store folder?
I don't know where it is, and I am running this in windows 10.
=======
would you be able to add a checkbox on each item in the list
and add a remove button?

it would make it easier for everyone to be able to remove all the unwanted entries..
instead of the 2 manual methods you mentioned..

yet-another-average-joe, 2019-09-07:
The history file path should be : C:\Users\abc\.data-store\marlin-config.json

At least with Windows, this MRU list is useless, as items cannot be clicked : it could be removed... Same for the "new" button.

madmax2, 2019-09-08:
okay i found the folder and deleted that file etc..

So is it possible to add some sort of delete button or clear all button to the start up main window of marlin conf so it is easier and quicker to delete either one or all items on that list?

The program should not make us know what file and where to find it to delete the items on that list..
that is not very user friendly.