akaJes/marlin-config #106: snoretoast - windows notification

madmax2, 2019-09-09:
why am I getting notifications about marlinconf via the app snoretoast ?
https://i.imgur.com/KbZnagG.png

how can i uninstall this unwanted application?
and how did it get installed with marlinconf?

akaJes, 2019-09-09:
Hi, this is a part of notification component
https://www.npmjs.com/package/node-notifier
for windows console notification used this sender
https://github.com/nels-o/toaster

<https://www.npmjs.com/package/node-notifier>

madmax2, 2019-09-09:
@akaJes why is it notifying about the marlin conf message?
I don't even know what message is about..

And how can I remove this snoretoast?
it is constantly popping up in my notification with that marlin message..

I don't even know how it got installed in the first place.