akaJes/marlin-config #110: SSL certificate error

kolorete78, 2020-03-14:
Hello @akaJes, after installing ubuntu 20.04, SSL get updated and require CipherString = DEFAULT@SECLEVEL=2

For that reaseon I' m unable to run mct as I did in previous versions.

Any idea on how to solve this problem ... is there any place during the installation process that ask to create new keys?

`pablo@rafiki:/home$ mct
_tls_common.js:135
      c.context.setCert(cert);
                ^

Error: error:140AB18F:SSL routines:SSL_CTX_use_certificate:ee key too small
    at Object.createSecureContext (_tls_common.js:135:17)
    at Server (_tls_wrap.js:873:27)
    at new Server (https.js:62:14)
    at Object.Server (https.js:44:41)
    at Object.<anonymous> (/usr/local/lib/node_modules/marlin-conf/app/server.js:34:23)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
`

kolorete78, 2020-03-14:
also tried /usr/local/lib/node_modules/marlin-conf/sslcert/gen script ... it run correcrly but problem persist ...

niai, 2020-09-04:
You can use the certs here https://github.com/abloom/marlin-config/tree/master/sslcert

just run 
`sudo mv sslcert/* /usr/local/lib/node_modules/marlin-conf/sslcert/`